rlogiacco/CircularBuffer #2: count should not go below 0

eto-, 2017-08-08:
I would add a check for shift & pop that before doing anything check if the fifo is non empty. Otherwise counts gets negative and the pointers get unaligned.

Something like:
`if (!count) return *head;`
before line 69 and 79.

rlogiacco, 2017-08-10:
Actually, my thinking was you shouldn't call either `shift` or `pop` unless `size` is greater than `0` or `isEmpty` is `false`...
I know the watch condition would improve in robustness, but it will also decrease speed. On the other hand if used unprotected they'll bring the whole buffer in an unpredictable and unrecoverable state...

Another reason why I didn't go for the suggested solution is you are actually returning a wrong value: `head` will *not* be empty, so once you reach the head of the buffer you can keep popping out the same element indefinitely... 

Which value do you believe should be returned in case you try to `pop` or `shift` from an empty buffer?
I couldn't find a way to properly mark it as an error...

eto-, 2017-08-10:
It is clear that the user should check the size, but if he doesn't?

Leaving the library in an unrecoverable state is IMHO the worst choice
since it may result in a condition hard to debug.
You can generate a hard error (division by zero, access to *0, as you
prefer) that at least crashes the system where the problem arises (and not
much later when the user try to use again the object).

Returning the wrong value is behind your control: you are already doing.
And even the [] operator may return wrong values if the index is outside
the size.
Since there are no exceptions you can not handle this.

An alternative you can use a different syntax that returns a bolean
(valid/unvalid) and write the value (if present) in a reference passed as
argument.
But I would guess most user would not check the status.

~ar

rlogiacco, 2017-08-10:
I like the hard error option and I agree about the `[]` operator: would you mind submitting a pull request? This way you'll get recognized as a contributor and get the credits you deserve ;-)

eto-, 2017-08-11:
Done

~ar

rlogiacco, 2017-08-20:
Fixed by pull request  #3

closed 2017-08-20 by rlogiacco