rlogiacco/CircularBuffer #5: License issue / question

5N44P, 2017-09-07:
Hi, 
First of all i want to thank you for this amazing library. I'm using it for a project and it's just perfect! I really love it!

I have only a question about licensing. It seems that there is a conflict between three different tipe of licenses.

In the [license file](https://github.com/rlogiacco/CircularBuffer/blob/4fb4a64/LICENSE) it is stated that the repo is under Apache-2.0 license:
> Apache License
> Version 2.0, January 2004
> http://www.apache.org/licenses/

But in the code file it seems to be under full copyright:
> CircularBuffer.tpp - Circular buffer library for Arduino.
> Copyright (c) 2017 Roberto Lo Giacco.  **All right reserved.**

And soon after under LGPL 2.3 or later: 
> This library is free software; you can redistribute it and/or
>  modify it under the terms of the GNU Lesser General Public
>  License as published by the Free Software Foundation; either
>  version 2.1 of the License, or (at your option) any later version.

So i wanted to ask you to clarify this, is the library under full copyright, Apache 2.0 or LGPL 2.3?

Thank you!! 😃

rlogiacco, 2017-09-07:
Thanks for noticing: I will fix that soon. BTW, may I ask why are you interested in licensing?

5N44P, 2017-09-07:
Actually... for no reason. 
It just caught my eye when i was trying to understand how the library works on the inside (without success 😂 😂 templates are a new thing to me, but definitely cool)
The final license won't make any difference for my project, as long as i can use it and link it in my (future) repo.

rlogiacco, 2017-09-07:
License has been updated here and will be available with next release 😉

closed 2017-09-07 by rlogiacco