rlogiacco/CircularBuffer #6: short = byte?

fdedraco, 2017-10-13:
is short type on your library equate to byte type on arduino?

as in 8 unsigned bit

rlogiacco, 2017-10-14:
I don't understand your question. The library is capable of handling both `short` and `byte` types, but they are not the same: a short is a 16-bit data-type ranging from `-32,768` to `32,767`, a byte is an 8-bit data-type ranging from `-128` to `127`.
The library does nothing regarding data-type manipulation.

fdedraco, 2017-10-21:
so i can use 
`CircularBuffer<unsigned char,100> data;`
 
for 8 bit unsigned number from `0-255`?

i'm confused on how you describe short and char use same amount of memory.
i guess i'll just try and see if something break

rlogiacco, 2017-12-06:
Don't be confused: you can use any data type you want.

As an example, I like to use `uint8_t` for unsigned 8 bits numbers, but that is exactly the same as `byte` or `unsigned char`.

I prefer the following because that helps me avoid confusion about the memory footprint: `uint8_t`, `uint16_t`, `uint32_t`, and `uint64_t` (unsigned) along with `int8_t`, `int16_t`, `int32_t`, and `int64_t` (signed).

Also you can use any other data type you like, including objects, structs or arrays... There's no limit on that front.

But remember, a `short` is **not** an 8 bit integer in Arduino world, that is 2 bytes, and as such it is the same type as `int`.

https://www.arduino.cc/reference/en/#variables
https://learn.sparkfun.com/tutorials/data-types-in-arduino
https://www.tutorialspoint.com/arduino/arduino_data_types.htm

closed 2017-12-06 by rlogiacco

fdedraco, 2017-12-06:
thanks, tried it and nothing broken, i forgot to report.
and thanks for the fix in README.
i was afraid if it somehow mutate the data entered if i used the wrong type.

> The library does nothing regarding data-type manipulation

is the confirmation i need btw