rlogiacco/CircularBuffer #9: Is the memset in clear() necessary?

cmooney3, 2018-04-11:
Hello!

I've been working on a project using this library and noticed that the clear() function seemed to be taking an inordinately long time on my microcontroller.  I did a little digging, and the culprit seems to be this line below, where you use memset() to erase all the data in the buffer:

https://github.com/rlogiacco/CircularBuffer/blob/473845ecc5295fb29629ed67b61db6b66023c618/CircularBuffer.tpp#L134

Since that function sets the head, tail, and count variables, my suspicion was that this memset isn't actually required.  The constructors seem to confirm this since they don't memset the buffer on creation (unless it gets initialized to 0 some other way I'm not seeing).

I went ahead and tested this out and everything seems to work great on my project at least, so I wanted to drop you a message and let you know in case this is a nice low-hanging fruit speedup.  If I'm wrong and that is important, then that'd be good to know too so I don't end up hitting some sneaky bug later on. 

Love the library, btw!  Thanks for writing it, it's saved me a lot of time

Thanks, Charlie

rlogiacco, 2018-04-22:
I've taken your advice and actually turned the `memset` switchable via a new, optional, parameter to the `clear()` function named _sweep_: it's active by default, for backward compatibility and as it might help recover unused space (under certain circumstances),but you can now provide a `false` value to obtain maximum performance.
The version has not been released yet, I'll wait for your confirmation it might be a valid approach.

tonton81, 2018-04-22:
if head and tail positions are reset, or a read/pop is done, theres no point to discard data in that slot as its not included in the queue, and further writes will override it anyways

adding memset is pretty redundant, but not necessary

rlogiacco, 2018-04-22:
if we all agree there is no value in having that clear up, I will remove it in the next version...

cmooney3, 2018-04-22:
Awesome, thanks for looking into it so quick :)

rlogiacco, 2018-04-24:
I've removed the `memset` call: in a couple of days I will release the new version for the plugin manager

commit 0c89fdab (2018-04-22): fix #9
closed 2018-04-24 by rlogiacco