rlogiacco/CircularBuffer #10: Can't use a CircularBuffer of structs?

schweini, 2018-05-07:
I'm new to structs, and it probably simply isn't supported, but I'm trying to use a CircularBuffer for some button-press structs, and get the error:

> no matching function for call to 'CircularBuffer<volatile buttonPresses, 10u>::push(buttonPress*)'

my code is:

```
typedef struct buttonPresses
{
  byte button;
  int time; 
} buttonPress;

#include <CircularBuffer.h>
#define CIRCULAR_BUFFER_INT_SAFE
CircularBuffer<volatile buttonPress, 10> pressesBuffer;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

  buttonPress foo = { 123, millis() };
  pressesBuffer.push( &foo );

}
```

rlogiacco, 2018-05-07:
This is a support request: please refer to the forum mentioned in the README and re-post your case over there.
Thanks

closed 2018-05-07 by rlogiacco

schweini, 2018-05-07:
> @schweini <https://github.com/schweini> I'm willing to help, but I also
> want to collect support requests in a different place: post your question
> over there and I'll try to help you out.

Thank you for your kind offer, and open sourcing your library!

But I switched to another library that does something similar, so I'm good.

And all I wanted to point out is that it would reduce user frustration a
bit if the docs would point out that only certain data types are supported.
Don't know if that's a "bug" or a "feature request".

rlogiacco, 2018-05-08:
Happy to know you solved your issue.

rlogiacco, 2018-05-08:
BTW, there are two examples in the examples folder describing how to use the library with pointers (your case) and structs: your "bug" vs "feature request" was actually just "where can I find an example usage of the library with structs?".
And to be 100% clear, the library works OK with [structs](https://github.com/rlogiacco/CircularBuffer/blob/master/examples/Struct/Struct.ino) and [pointers](https://github.com/rlogiacco/CircularBuffer/blob/master/examples/Object/Object.ino) (to either structs or objects) and would have also worked for you if you had declared the buffer properly:

```cpp
CircularBuffer<volatile buttonPress*, 10> pressesBuffer;
```

Sadly, as this was not posted on the forum, the next one having your same issue will have to dig into the defect tracking.

rlogiacco, 2018-05-08:
And to wrap it up, please consider the author's frustration when users don't read the examples and documentation he wrote to help them out...