rlogiacco/CircularBuffer #11: I need the address of the array. How can I get the address of an array?

highspeeder, 2018-08-10:
I set a pointer to fetch the address value of the buffer, and put the address value of the buffer. However, I get the following error:
"The expression must be an lvalue or function specifier"

in code:
```
CircularBuffer<char, 10> chars;
char* pGet = NULL;

void loop()
{
    pGet = &chars[0];
}
```
How can I get the address value of the buffer?

rlogiacco, 2018-08-10:
Your last attempt returns a pointer to the CircularBuffer, which is not a char.

If you need a pointer to an element you can use

```
char c = chars[0];
pGet = &c;
```

If you intend to get access to the internal structure that is protected from external access: being a circular buffer there is no guarantee your _first_ item precedes the _second_ item in the internal structure...

rlogiacco, 2018-08-10:
By re-reading the title of the question I believe there's a misunderstanding: a circular buffer is a data structure, like an array or a linked list, but it's not an array and you cannot "cast" the circular buffer to an array.
It internally uses an array, but that's an implementation detail which this library tries to hide as much as possible: otherwise why use the library...
You can create an array with the content of the circular buffer though, by for-looping over the buffer and populating a properly sized array: I wonder if the circular buffer is the proper data structure in this case though

closed 2018-08-17 by highspeeder