rlogiacco/CircularBuffer #19: Can you explain how to add strings (char *) to Circbuffer

geetee24, 2019-02-10:
Hi

I am reading a char* (like "See cat play") into RAM and I want to save it into your CircB but the readme does not explain how to.  Can you tell me please?

Of course, when circb gets full, I want it to free the RAM when it dumps the oldest entry.

Thank you.

rlogiacco, 2019-02-10:
Considering this question it's probably better fit for a StackExchange entry, a quick answer is probably to read one char at a time and push each char into a buffer declared as operating on `char`
`CircularBuffer<char, 32> buffer;`

closed 2019-02-10 by rlogiacco

geetee24, 2019-02-11:
Thank you but I need to be able to retrieve a string at a time can your
library work with strings

rlogiacco, 2019-02-11:
The library can work with strings, with each string occupying a slot of the buffer: 
`CircularBuffer<char*, 32> buffer;`

This buffer will store a maximum of 32 strings, each one represented as an array of chars: no limit is imposed on the length of the strings (either combined or not), with the _wrap around_ occurring when you try to push the 33th string.

If you want to operate on the String object, the library can operate on objects and structures as well (see examples).

I don't know what your goal is, but my gut feeling is you are on the wrong track here.

proddy, 2019-02-11:
@geetee24 I'm not sure either what you're trying to do. If you're planning on storing ordered strings in PROGMEM have a look C++ containers like [std::queue](https://en.cppreference.com/w/cpp/container/queue), or vector and list.

I use Roberto's code because its small and super efficient, but if speed and memory is not an issue for you also look at using containers.

geetee24, 2019-02-11:
Let's say I am reading a series of strings coming in from an IOT device and
I simply want to save each of those strings into a circular buffer for a
later inspection

rlogiacco, 2019-02-11:
The simplest way to achieve that is by storing `String` objects in the CircularBuffer structure, definitely not the most efficient though.
If you are willing to deal yourself with the null termination handling when necessary, you can go for storing `char*` in the CircularBuffer, making the solution a little less memory hungry.

Please note the memory consumption is unknown at compile time. in both cases: longer the char sequences, more the memory required to store them.

Have I forgot something @proddy ?

proddy, 2019-02-11:
That'll work if you want to be safe and preallocate some RAM. If you don't care use a vector/list/queue which dynamically allocates memory when a new record is added. It is costly though. Also I would avoid using Strings at all costs 'cos their plain ole nasty. If the idea is to read in strings say from UART stream or MQTT messages then I would create a struct as you'll probably want to store additional info like string length and timestamps.

geetee24, 2019-02-11:
I agree with you.  Never use "Strings".    just want to know if the Cicr
lib can handle  a series of "char* and FREE them up when the CircB is full ?

proddy, 2019-02-11:
not sure what you mean about 'free'. The allocated memory is already reserved so there's nothing to free up if you're referring to memory. It's a circular buffer so it'll just go round and round. As long as you correctly null-terminate the char* you won't be picking up stray data from unused memory.

geetee24, 2019-02-11:
the circ buffer had a feature that if you add more elements than it can
hold, it will remove the oldest element.

proddy, 2019-02-11:
that's correct, the entry is overwritten

geetee24, 2019-02-11:
the question I want to know, is does the library FREE (release up) the
memory that the char* used?

proddy, 2019-02-11:
no it doesn't! This is what Roberto explained earlier. When you initialize the buffer the compiler will allocate the memory. That what makes it zippy fast

rlogiacco, 2019-02-11:
> the question I want to know, is does the library FREE (release up) the memory that the char* used?

The `char*` is a pointer, internal to the library, as such the pointer is never deallocated, it is instead reused. On the other end, the sequence of characters composing the stored string, is simply left to your handling: as you and only you know how you got that sequence of characters, only you know how to properly dispose of it, either deallocating or leaving it behind.

Now the question is, how would you dispose of such a sequence of characters? If you were reading some stuff from the serial input, would you call a `delete` or `memfree` once you are done with those? If the answer is no, then you don't have to do anything specific to free up the memory, otherwise use the exact same approach you would use normally: simply do that right after a `pop` or a ` unshift` operation.

geetee24, 2019-02-12:
But I thought when the buffer circular was full search of a buffer its self
throughout the oldest entry and therefore I ask does the library or how
does the library free app that memory

rlogiacco, 2019-02-19:
You already got the answer you were looking for: as the buffer does not allocate memory it is not allowed to deallocate it, as simple as that. You, the user, are in charge of allocating/deallocating memory for objects on the heap.

How? Well, you check if you have any available space: if you don't, then pop and deallocate before pushing another one: the functions you are looking for are `pop`, `available` and `unshift`:

```cpp
your_type next_item = new your_type; // allocate memory either with new or malloc
if (available() == 0) {
  // buffer is full, dispose before adding
  your_type t = buffer.pop();
  delete t; // dispose of your heap accordingly to your allocation method: delete or free
}
buffer.unshift(next_item);