rlogiacco/CircularBuffer #21: Call to abort() is AVR-specific

danwahl, 2019-07-06:
Upgrading from an ATmega32u4 to ATSAMD21G board results in the following compiler error:
```
CircularBuffer.tpp:64:24: error: there are no arguments to 'abort' that depend on a template parameter, so a declaration of 'abort' must be available [-fpermissive]

  if (count <= 0) abort();
```
It seems like `abort()` is AVR-specific, and also [probably bad practice](https://forum.arduino.cc/index.php?topic=144756.0) to call directly. Recommend undoing the change in 86b33dc7dcd0480c044dd6dd4aff79e0313d4ceb as an easy fix.

rlogiacco, 2019-07-07:
Are you suggesting to replace `abort()` with `crash()`? Because that's what reverting 86b33dc would imply...

rlogiacco, 2019-07-07:
I am tempted to return to the original implementation which didn't have the check altogether: crashing the program is not nice, but also we cannot avoid returning something.
There is no reliable way to protect the user from bad usage of anything inside an Arduino or similar mcu...

closed 2019-07-07 by rlogiacco

rlogiacco, 2019-07-07:
@Erlkoenig90 I hope you agree with this solution. I wish we can start back working on your improvements and contributions: I had a tough period at work, but this might be a good time.

Now releasing `1.3.1` applying this fix

Erlkoenig90, 2019-07-07:
`abort()` is not AVR-specific, it's [Standard-C(++)](https://en.cppreference.com/w/c/program/abort). There's probably just an `#include <stdlib.h>` missing. The correct C++ header `<cstdlib>` probably doesn't work on AVR...

danwahl, 2019-07-07:
Thanks for the quick response. Looks like @Erlkoenig90 is right, but removing the call to `abort()` seems fine for now. Eventually, it might be best to change the interface to something like:
```c++
template<typename T, size_t S, typename IT>
bool CircularBuffer<T,S,IT>::shift(T &value) {
	if (count <= 0) return false;
	value= *head++;
	if (head >= buffer + capacity) {
		head = buffer;
	}
	count--;
	return true;
}
```

rlogiacco, 2019-07-07:
@danwahl the API change you are suggesting is not a viable alterantive: other than introducing a breaking change it would also make the `shift` and `pop` operations useless as there would be no way to access the shifted/popped element.

I already had a conversation with @Erlkoenig90 where I expressed my doubts regarding the use of `abort()`... I've never seen it used in any library

danwahl, 2019-07-07:
@rlogiacco understood re: breaking the API, but just to note, this would allow you to access the shifted/popped element through `value` (which is passed by reference). Also, it could be introduced as a feature with overloaded `shift` and `pop` functions. Anyway, just a thought - thanks for your work on the library.

commit 1fdec02c (2019-07-07): fixes #21 call to abort() is AVR-specific