rlogiacco/CircularBuffer #22: Where to put #define CIRCULAR_BUFFER_INT_SAFE

skleeschulte, 2019-09-20:
Being new to C/C++, I wonder where I should put `#define CIRCULAR_BUFFER_INT_SAFE`. In [README.md](https://github.com/rlogiacco/CircularBuffer/blob/1.3.1/README.md#interrupts) it is shown before variable declaration:

    #define CIRCULAR_BUFFER_INT_SAFE
    CircularBuffer<unsigned long, 10> timings;

In the [Interrupts.ino example](https://github.com/rlogiacco/CircularBuffer/blob/1.3.1/examples/Interrupts/Interrupts.ino#L1-L3) it is shown before the `#include` statement:

    #define CIRCULAR_BUFFER_INT_SAFE
    #include "CircularBuffer.h"
    CircularBuffer<unsigned long, 10> timings;

Does it matter?

proddy, 2019-09-20:
think it does, best put it before the #include to be safe

rlogiacco, 2019-09-20:
@proddy is definitely right, it goes before the import: anywhere after the import and it doesn’t work.

closed 2019-09-20 by rlogiacco