rlogiacco/CircularBuffer #26: Question: How can store CircularBuffer in PSRAM on ESP32?

debsahu, 2020-02-04:
I would like no not use the RAM on ESP32, but rather use PSRAM attached using SPI interface. Data can be stored directly on PSRAM using ps_malloc() as per [esp32-hal-psram.c](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-psram.c).

I declare `CircularBuffer<time_t, 2880> logTimeBuffer;` globally that stores stuff in regular RAM. So is there a way to call ps_malloc() to use the PSRAM?

There is a good reference on doing this for ArduinoJson [here](https://arduinojson.org/v6/how-to/use-external-ram-on-esp32/)

rlogiacco, 2020-02-04:
You can either change the source code for your case or use a macro switch to switch to a different implementation: I might consider accepting a pull request which uses the latter, but you need to test it also against AVR8 platforms as I don't want to break the library to support a corner case.

debsahu, 2020-02-06:
@rlogiacco I changed the array `T buffer[S]` to pointer `T buffer*` and used `malloc()` for all other boards and `heap_caps_malloc()` for ESP32 with PSRAM. I am not an expert in C/C++, but gave it a shot. I need help with why it is crashing.

rlogiacco, 2020-04-08:
Sorry, I'm not an expert on ESP32, but if you found a solution I'll be very willing to publish your solution.

debsahu, 2020-04-09:
Thank you will try that, my testing ESP32 unit with PSRAM is stuck at work (future me: due to a global pandemic). I will test it out next week.

rlogiacco, 2020-05-20:
closing this as we got no feedback from the OP

closed 2020-05-20 by rlogiacco