rlogiacco/CircularBuffer #27: Compilation error since v1.3.2

FrenchLab47, 2020-02-05:
Hello,

I have been using CircularBuffer for several months in my sketch, but since version 1.3.2 I get a compilation error.
Version 1.3.1 works perfectly.

I use it for three very simple int array

CircularBuffer <int, 3> stackpage1;
CircularBuffer <int, 3> stackpage2;
CircularBuffer <int, 6> stackpage;

Here is the message of the compilation, in Arduino IDE:

`In file included from D:\Documents_D\Arduino\libraries\CircularBuffer/CircularBuffer.h:147:0,
                 from sketch\010_global_includes.h:43,
                 from D:\Documents_D\Arduino\ESP_GRAPHOCLOCK_V1-431\ESP_GRAPHOCLOCK_V1-431.ino:7:

D:\Documents_D\Arduino\libraries\CircularBuffer/CircularBuffer.tpp: In instantiation of 'T CircularBuffer<T, S, IT>::operator[](IT) const [with T = int; unsigned int S = 3u; IT = unsigned char]':
sketch\024_Scroller_fnc.h:241:37:   required from here

D:\Documents_D\Arduino\libraries\CircularBuffer/CircularBuffer.tpp:96:12: error: invalid use of member function (did you forget the '()' ?)
  if (index >= size) return *tail;
            ^

D:\Documents_D\Arduino\libraries\CircularBuffer/CircularBuffer.tpp: In instantiation of 'T CircularBuffer<T, S, IT>::operator[](IT) const [with T = int; unsigned int S = 6u; IT = unsigned char]':
sketch\024_Scroller_fnc.h:269:36:   required from here

D:\Documents_D\Arduino\libraries\CircularBuffer/CircularBuffer.tpp:96:12: error: invalid use of member function (did you forget the '()' ?)`

Thanks for your help.

rlogiacco, 2020-02-05:
I believe the issue is caused by an error that was already fixed, but probably the released version still contains the bug.

I'm releasing version 1.3.3 right now, please update this if that doesn't fix the problem.

closed 2020-02-05 by rlogiacco

FrenchLab47, 2020-02-05:
I confirm that it works with version 1.3.3.
Thank you so much.

coogle, 2020-09-11:
I can also confirm this fixed the issue on my build with the same problem (upgrading from 1.3.2 to 1.3.3)

commit a6f5a376 (2020-02-05): releasing 1.3.3

fix #27