rlogiacco/CircularBuffer #28: Issue compiling on Arduino Nano 33 BLE

ShawnHymel, 2020-02-13:
When I try to compile/upload the CircularBuffer example, I get the following error:

```
CircularBuffer:3:1: error: 'CircularBuffer' does not name a type; did you mean 'CellularBase'?

 CircularBuffer<int, 400> buffer;

 ^~~~~~~~~~~~~~
```

It seems to work when I switch the board to something else (e.g. UNO), but it does not work with the Nano 33 BLE.

Any help would be appreciated.

Board: Arduino Nano 33 BLE
Arduino IDE 1.8.11
CircularBuffer Library 1.3.3

rlogiacco, 2020-02-13:
I don’t have a Nano 33 BLE, I can only suggest to check the library is correctly picked by the compiler among the inclusions...
If you are using the Arduino IDE enabling the compilation verbose output might help finding the issue

ShawnHymel, 2020-02-13:
Thanks for the quick response! I'm thinking it's an issue unique to the Nano 33 BLE and/or my machine (I've got the ARM compiler elsewhere on my path, but it doesn't look like Arduino is trying to use that one). I did a compiler with verbose, and it looks like Arduino is simply not finding the library. If you'd like to try it, you can just install the Nano 33 BLE board in the IDE and try compiling the example--no need for the actual hardware.

With the UNO selected and verbose, here's what I get for the "Detecting Libraries" step:

```
Detecting libraries used...
"C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10811 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\arduino-1.8.11\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\arduino-1.8.11\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\SGMUST~1\\AppData\\Local\\Temp\\arduino_build_796330\\sketch\\CircularBuffer.ino.cpp" -o nul
Alternatives for CircularBuffer.h: [CircularBuffer@1.3.3 SmartMatrix-master@3.2.0]
ResolveLibrary(CircularBuffer.h)
  -> candidates: [CircularBuffer@1.3.3 SmartMatrix-master@3.2.0]
"C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10811 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\arduino-1.8.11\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\arduino-1.8.11\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\sgmustadio\\Documents\\Arduino\\libraries\\CircularBuffer" "C:\\Users\\SGMUST~1\\AppData\\Local\\Temp\\arduino_build_796330\\sketch\\CircularBuffer.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10811 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\arduino-1.8.11\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\arduino-1.8.11\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\sgmustadio\\Documents\\Arduino\\libraries\\CircularBuffer" "C:\\Users\\SGMUST~1\\AppData\\Local\\Temp\\arduino_build_796330\\sketch\\CircularBuffer.ino.cpp" -o "C:\\Users\\SGMUST~1\\AppData\\Local\\Temp\\arduino_build_796330\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\arduino-1.8.11\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\SGMUST~1\\AppData\\Local\\Temp\\arduino_build_796330\\preproc\\ctags_target_for_gcc_minus_e.cpp"
```

With the Nano 33 BLE selected, here's what I get (notice that it doesn't seem to look for the CircularBuffer at all):

```
Using core 'arduino' from platform in folder: C:\Users\sgmustadio\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4
Detecting libraries used...
"C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10811 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\sgmustadio\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\SGMUST~1\\AppData\\Local\\Temp\\arduino_build_796330\\sketch\\CircularBuffer.ino.cpp" -o nul
```

rlogiacco, 2020-02-13:
I don’t even use the Arduino IDE for my development, so...

Is it possible the MBed libraries require some specific detail to be picked up? If I remember correctly the Arduino IDE can show the list of libraries you have included in your project: is Circular Buffer in that list?
Sorry for not being of greater help, but it clearly seems like a tool chain configuration issue

ShawnHymel, 2020-02-21:
No worries. It does seem like a toolchain issue. I can't find it exactly, but I can keep digging. I'm happy to close out the issue, knowing it's specific to one particular Arduino board (that seems to be having lots of issues with other libraries, anyway).

rlogiacco, 2020-02-22:
Feel free to reopen it if you get additional details leading to something wrong in our code.

Knar33, 2020-04-17:
I'm having this issue with the same board, did you ever manage to solve it @ShawnHymel ?

ShawnHymel, 2020-04-17:
@Knar33 no dice :( I ended up writing my own simple ring buffer instead.

rlogiacco, 2020-04-17:
That's a pity... do you mind to share your code so we can try to give that a comparison? May I suggest you give a try at copying the library files into your project and see if the issue has anything to do with the code? It might be related to some metafile we aren't aware of, rather than to the code itself...

Knar33, 2020-04-17:
The reference came from a library I was trying to use: https://github.com/cesarvandevelde/HerkulexServo
I did try copying and referencing everything locally with no luck. It almost seems like the tpp file is being ignored by the compiler but I don't know enough about c++ to really dig into it. This isn't the first compatibility issue I've ran into with the Nano 33, the board itself seems to be the common problem.

ShawnHymel, 2020-04-18:
Agreed...I like the idea of the Nano 33, but it's proven to have massive compatibility issues with many Arduino libraries, making it really difficult to work with.

RicoWilson, 2021-12-23:
I'm using the CircularBuffer library on a nano BLE sense. I had the same problem. To remove my code from the debug process, I tried to compile the logging example with the board set to a Uno (successful). I then set the board to nano ble sense and got the 'does not name a type' error. I have a hunch that the keyword CircularBuffer is used somewhere in the code that arduino-fies the BLE sense. I copied the library to a local library named RcircularBuffer (to avoid my hypothesized name conflict) and changed all instances of Circular to Rcircular. I changed the CIRCULAR_DEBUG definitions to RCIRCULAR_DEBUG definitions, though that may have been unnecessary. I changed my code to reference RcircularBuffer and was able to compile and load my code. 

I offer this as debugging since resolving the name conflict in a civilized way is undoubtedly messy, but for anyone wanting to create a local library, this solution seems to work.  Thanks, Roberto, for a great library!

Please be kind, this is my first comment on github...

digamesystems, 2021-12-23:
C++ allows the use of namespaces. This might be a job for a CircularBuffer namespace. (Reference for interested folks: https://www.geeksforgeeks.org/namespace-in-c/)

They're designed exactly for this sort of thing. 

Ex from the above ref: 

```
// Creating namespaces
#include <iostream>
using namespace std;
namespace ns1
{
	int value() { return 5; }
}
namespace ns2
{
	const double x = 100;
	double value() { return 2*x; }
}

int main()
{
	// Access value function within ns1
	cout << ns1::value() << '\n';

	// Access value function within ns2
	cout << ns2::value() << '\n';

	// Access variable x directly
	cout << ns2::x << '\n';	

	return 0;
}

```

rsalvati, 2022-11-13:
Thanks to the advice of @RicoWilson I found an internal library called "CircularBuffer.h", so it was a name conflict. I solved specifying the path inside the include function:

`#include <C:\Users\Raffaele\Documents\Arduino\libraries\CircularBuffer\CircularBuffer.h>`

Also, I had to specify the path of the CircularBuffer.tpp inside the CircularBuffer.h script (placed at the end).
I just tried the library on my Nano 33 BLE Sense board and it works fine.

rlogiacco, 2023-02-08:
Can some of you please check the solution currently posted on the `fix/library-include` branch? I changed the name of the header file to CircularBuffer.hpp, which apparently I should have used since the very beginning as this is a C++ library.

I wonder if that itself is a valid fix.

Thanks a million to whomever dedicates the time to run this check for me.

closed 2023-12-15 by github-actions[bot]