rlogiacco/CircularBuffer #34: error: 'CircularBuffer' does not name a type; did you mean 'CellularBase'?

andreamartini, 2020-05-20:
Hi,
i'm trying to run example CircularBuffer.ino from example folder of CircularBuffer Library. I installed 1.3.3 version with Libreries Manager, and i'm using arduino ide 1.8.12 
I'm struggling to understand and  to solve the following compiling error:

> error: 'CircularBuffer' does not name a type; did you mean 'CellularBase'?

Following, i report the complete error message:
```
CircularBuffer:3:1: error: 'CircularBuffer' does not name a type; did you mean 'CellularBase'?

 CircularBuffer<int, 400> buffer;

 ^~~~~~~~~~~~~~

 CellularBase

CircularBuffer:5:15: error: 'long unsigned int time' redeclared as different kind of symbol

 unsigned long time = 0;

               ^~~~

In file included from c:\users\andrea\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\string.h:10:0,

                 from C:\Users\Andrea\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/Callback.h:20,

                 from C:\Users\Andrea\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Thread.h:29,

                 from C:\Users\Andrea\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/rtos.h:28,

                 from C:\Users\Andrea\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/mbed.h:23,

                 from C:\Users\Andrea\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Arduino.h:32,

                 from C:\Users\Andrea\AppData\Local\Temp\arduino_build_776870\sketch\CircularBuffer.ino.cpp:1:

c:\users\andrea\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\time.h:59:11: note: previous declaration 'time_t time(time_t*)'

 time_t    _EXFUN(time,     (time_t *_timer));

           ^

C:\Users\Andrea\Documents\Arduino\libraries\CircularBuffer\examples\CircularBuffer\CircularBuffer.ino: In function 'void setup()':

CircularBuffer:12:16: error: assignment of function 'time_t time(time_t*)'

  time = millis();

                ^

CircularBuffer:12:16: error: cannot convert 'long unsigned int' to 'time_t(time_t*) {aka long long int(long long int*)}' in assignment

C:\Users\Andrea\Documents\Arduino\libraries\CircularBuffer\examples\CircularBuffer\CircularBuffer.ino: In function 'void loop()':

CircularBuffer:18:2: error: 'buffer' was not declared in this scope

  buffer.push(reading);

  ^~~~~~

C:\Users\Andrea\Documents\Arduino\libraries\CircularBuffer\examples\CircularBuffer\CircularBuffer.ino:18:2: note: suggested alternative: 'buffer_t'

  buffer.push(reading);

  ^~~~~~

  buffer_t

CircularBuffer:20:15: error: invalid operands of types 'long unsigned int' and 'time_t(time_t*) {aka long long int(long long int*)}' to binary 'operator-'

  if (millis() - time >= 500) {

      ~~~~~~~~~^~~~~~

CircularBuffer:21:17: error: assignment of function 'time_t time(time_t*)'

   time = millis();

                 ^

CircularBuffer:21:17: error: cannot convert 'long unsigned int' to 'time_t(time_t*) {aka long long int(long long int*)}' in assignment

CircularBuffer:24:19: error: decltype evaluates to '<type error>', which is not a class or enumeration type

   using index_t = decltype(buffer)::index_t;

                   ^~~~~~~~

CircularBuffer:25:8: error: 'index_t' was not declared in this scope

   for (index_t i = 0; i < buffer.size(); i++) {

        ^~~~~~~

C:\Users\Andrea\Documents\Arduino\libraries\CircularBuffer\examples\CircularBuffer\CircularBuffer.ino:25:8: note: suggested alternative: 'index'

   for (index_t i = 0; i < buffer.size(); i++) {

        ^~~~~~~

        index

CircularBuffer:25:23: error: 'i' was not declared in this scope

   for (index_t i = 0; i < buffer.size(); i++) {

                       ^

exit status 1
'CircularBuffer' does not name a type; did you mean 'CellularBase'?

```

Any suggestions are welcome.
Thank you in advance.

Kind Regards

andreamartini, 2020-05-20:
Hi,
i'm sorry about submitting my previous post. There is the same issue on:
[](https://github.com/rlogiacco/CircularBuffer/issues/28)
I'm using arduino nano ble 33 sense too, and i get the same compile error reported at the link above. I tried also other CircularBuffer version 1.3.1 and 1.3.2 with no success.
Thank you
Kind Regards

andreamartini, 2020-05-20:
Sorry,
following link i referred to:
https://github.com/rlogiacco/CircularBuffer/issues/28[](url)

rlogiacco, 2020-05-21:
Are you also using a Nano BLE board?

andreamartini, 2020-05-21:
Hi,
yes, i'm using Nano BLE 33 Sense Board.

rlogiacco, 2020-05-22:
Then this is a duplicate of #28 : I'm sorry, but it doesn't seem the issue depends on the library but rather on the toolchain.

closed 2020-05-22 by rlogiacco