rlogiacco/CircularBuffer #37: Fills the whole buffer with first value

maxholgasson, 2020-11-01:
I'm having the problem that when I push the first value to the buffer, the whole buffer consists of this value instead of only at one position
As an example

byte ID = 3;
..
buffer_1.push(ID);

gives 
buffer_1[0]->3
buffer_1[1]->3
buffer_1[2]->3

`CircularBuffer<byte, 3> buffer_1;`

rlogiacco, 2020-11-02:
Please read the documentation, you are misusing the library: if you pushed
one value only you cannot read more than one value from the buffer.
I have documented the operator specifically for what you are reporting as
an issue.

maxholgasson, 2020-11-02:
Is it what you're doing with 
>for (index_t i = 0; i < buffer.size(); i++) 
?

rlogiacco, 2020-11-02:
In the README (relevant part in **bold**)

> ⚠ **Reading data beyond the actual buffer size has an undefined behaviour** and is user's responsibility to prevent such boundary violations using the additional operations listed in the next section. The library will behave differently depending on the data type and allocation method, but you can safely assume your program will crash if you don't watch your steps.

So, yes, you need to iterate only over the valid cells in the buffer: anything beyond that might crash your entire firmware as you are accessing invalid memory areas. As there is no way to report an exceptional condition and I wanted to keep the library as lightweight as possible, I count on documentation and proper usage.

closed 2020-11-02 by rlogiacco

maxholgasson, 2020-11-02:
Ok, eitherway it would be nice to have an example for a beginner
>hat might crash your entire firmware 
That's bad, thanks for the info I'll look for another library

rlogiacco, 2020-11-02:
You are welcome. Let me know if you decide to come back :-)