rlogiacco/CircularBuffer #40: Circular Buffer requires declarion as ino global

AbbottHMG, 2021-01-03:
Roberto has been helping me with my use of CB. After many tries (and great support from Roberto) I located the issue was  I was having trying to use CB within a class. It seems that CB only links properly if declared globally in the .ino file. When I used extren  in the class to reference CB all worked (linked) well.
I'm not sure that a code change is necessary, but it would be good to  have a heads up in the documentation. It would have saved Roberto a lot of aggravation time with my issue.

rlogiacco, 2021-01-04:
Can you provide a failing code example, please?

Thank you

AbbottHMG, 2021-01-04:
[CBTest.zip](https://github.com/rlogiacco/CircularBuffer/files/5765753/CBTest.zip)
CBtest compiles fine...with the externs. Comment the exteerns in CBRunner.h and static declarations in ino and un-comment the static  definitions in CBRunner.h and the errors appear in output.

Error linking for board ATmega2560 (Mega 2560) (Arduino Mega)
ccOXq4De.ltrans0.ltrans.o*: In function CBRunnerClass::init()
   (.text+0x5e2): undefined reference to CBRunnerClass::tasks
   (.text+0x5e4): undefined reference to CBRunnerClass::tasks
   (.text+0x606): undefined reference to CBRunnerClass::CBuffer
Build failed for project 'CBTest'
   (.text+0x608): undefined reference to CBRunnerClass::CBuffer
 
collect2.exe*: error: ld returned 1 exit status

rlogiacco, 2021-01-06:
Your issue is with the language, this library has nothing to do with the reported error, with the proof being the lines you reported yourself:

```
(.text+0x5e4): undefined reference to CBRunnerClass::tasks
(.text+0x606): undefined reference to CBRunnerClass::CBuffer
```

As you can see, the compilation error is reported **both** for the CircularBuffer instance **and** for the Task array. The issue you are having is with the static members inside the class: remove the `static` keyword on both attributes and functions in the `CBRunnerClass` and everything goes back to _compilation success_.

As a counter proof, feel free to comment any reference to the library and leave only your static array definition to appreciate the very same error being reported solely for the missing reference to `CBRunnerClass::tasks`.

The `CircularBuffer` works perfectly fine as a member attribute of a class, either static or not.

closed 2021-01-06 by rlogiacco

rlogiacco, 2021-01-06:
Add these two lines to your `CBRunner.cpp` file and remove the unecessary cross references:

```cpp
TaskClass CBRunnerClass::tasks[] = {};
CircularBuffer<TaskClass*, 5> CBRunnerClass::CBuffer;
```

AbbottHMG, 2021-01-07:
Thanks you ... again, sorry for the bother.
Abbott