rlogiacco/CircularBuffer #45: Allow using buffers as instance members

murar8, 2021-02-18:
First of all, I would like to thank you for the nice work.

Right now it looks like it is impossible to use a circular buffer as an instance member since you are using the `constexpr` constructor. I am by no means a c++ expert but it does not look like there is any technical limitation that impedes doing it so maybe we could revert to using the normal constructor?

Minimal example:

``` cpp
#include <CircularBuffer.h>

class Foo {
    CircularBuffer<int, 400> buffer_;
};

void setup() {
  Foo foo1 = Foo();
  Foo foo2 = Foo();
}

void loop() {
}
```

Compiler error:

```
C:\Users\lnzmr\OneDrive\Documenti\Arduino\CircularBuffer_Test\CircularBuffer_Test.ino: In function 'void setup()':
CircularBuffer_Test:8:18: error: use of deleted function 'Foo::Foo(Foo&&)'
   Foo foo1 = Foo();
                  ^
C:\Users\lnzmr\OneDrive\Documenti\Arduino\CircularBuffer_Test\CircularBuffer_Test.ino:3:7: note: 'Foo::Foo(Foo&&)' is implicitly deleted because the default definition would be ill-formed:
 class Foo {
       ^~~
CircularBuffer_Test:3:7: error: use of deleted function 'CircularBuffer<T, S, IT>::CircularBuffer(CircularBuffer<T, S, IT>&&) [with T = int; unsigned int S = 400; IT = unsigned int]'
In file included from C:\Users\lnzmr\OneDrive\Documenti\Arduino\CircularBuffer_Test\CircularBuffer_Test.ino:1:0:
C:\Users\lnzmr\OneDrive\Documenti\Arduino\libraries\CircularBuffer/CircularBuffer.h:59:2: note: declared here
  CircularBuffer(CircularBuffer&&) = delete;
  ^~~~~~~~~~~~~~
CircularBuffer_Test:9:18: error: use of deleted function 'Foo::Foo(Foo&&)'
   Foo foo2 = Foo();
                  ^
Uso la libreria CircularBuffer alla versione 1.3.3 nella cartella: C:\Users\lnzmr\OneDrive\Documenti\Arduino\libraries\CircularBuffer 
exit status 1
use of deleted function 'Foo::Foo(Foo&&)'
```

Erlkoenig90, 2021-02-19:
This has nothing to do with `constexpr`, but because the copy/move constructors/assignment operators are deleted. They could be implemented, but since they're bound to be inefficient, this wouldn't be very useful for a microcontroller.

This line
```c++
Foo foo1 = Foo();
```
Creates a temporary instance of `Foo`, and then attempts to create a new instance (called `foo1`) and move or copy the temporary instance's data into that. Since `CircularBuffer` can't be moved/copied, `Foo` can't be either, so you get an error. Even if you could copy/move CircularBuffer, it would be inefficient to create a temporary instance, then copy/move it, and then destroy it right after that. You can simply do:
```c++
Foo foo1;
```
This will simply create one instance of `Foo`, without copying/moving/temporary instances. This should work fine. If you need to pass parameters to `Foo`'s constructor, do it as
```c++
Foo foo1 (a, b, c);
```
The pattern of creating instances like `Foo foo1 = Foo();` doesn't work well in C++ and should be avoided.

murar8, 2021-02-19:
Thank you, that was a very clear explanation and it works perfectly, sorry for wasting your time.

closed 2021-02-19 by murar8