rlogiacco/CircularBuffer #46: 'CircularBuffer<T, S, IT>::buffer' has incomplete type

MohimenAlMahaini, 2021-03-16:
In CircularBuffer.h there is a problem with T buffer[s];
'CircularBuffer<T, S, IT>::buffer' has incomplete type

rlogiacco, 2021-03-17:
This hardly makes any sense.
First of all, take a step back and try the provided examples, verifying if the error is still there. 
If the error is not occurring with the examples, please consider double-checking the code producing the error: you should agree it is more probable you did something wrong rather than _"there's a problem within the library"_. 
If, after those checks, you are still wondering what's wrong with the use you are making of the library and you are seeking help, you should provide the code which generates such an error along with any detail necessary to reproduce it, like the toolchain you are using for compilation.

By the tiny bit of information you have provided though, I'm 99% convinced you are misusing the library, probably trying to use it as if it is an array.

MohimenAlMahaini, 2021-03-17:
Sorry probably I'm wrongly implementing the library, 

here's the code:

`#include <CircularBuffer.h>

CircularBuffer<struct RX, 15> buf_BAPrx;
CircularBuffer<struct RX, 15> buf_BAPTx;

RX B;

void test()
{
    RX B; /* ERORR: incomplete type is not allowed */

}
`
"The code is trying to created an instance of an object on the stack. This object's type is only forward declared but no definition is available to the compiler at that time.

You need to make the definition of this type available to the compiler before it has to create an instance of this type." selalerer

closed 2021-03-17 by MohimenAlMahaini