rlogiacco/CircularBuffer #47: Can't use library within a class.

gnalbandian, 2021-03-31:
Hi.
I am trying to use this library as an object within a class and I am getting the following error:

` error: use of deleted function 'constexpr CircularBuffer<timeTempStruct, 10u>::CircularBuffer(const CircularBuffer<timeTempStruct, 10u>&)'`

The class is globally instanced as:

`className objectName[8] = className (constructor arguments);`

Any tip on how to make this work?
Thanks!

Regards!
Gonzalo

gnalbandian, 2021-04-01:
Solved by not using copy constructor when instantiating the class.

closed 2021-04-01 by gnalbandian

dzalf, 2024-02-13:
@gnalbandian Sorry for reviving this thread so late. How exactly did you manage to solve this?

I am facing exactly the same issue:

[image]