rlogiacco/CircularBuffer #50: Array-like indexed write operation?

drws, 2021-09-11:
`CircularBuffer` currently only supports array-like reading operation (`data = buffer[0]`), but not a writing one (`buffer[0] = data`). I know the library tries to protect internal buffers as much as possible, but this limitation is effectively making content of elements in CircularBuffers read-only. To modify a value in the buffer, one needs to reshift half of it (on average) or am I missing something?

rlogiacco, 2021-09-12:
If you submit a pull request I will evaluate if it fits into the scope of the library, but generally, I don't see the use for such a direct manipulation because in a circular buffer you can associate a meaning only to the head or the tail elements: any modification of anything in between is at high risk.

So I guess you'll have to also convince me it makes sense for those direct manipulation operations to have a sense within the scope of a circular buffer... 😄

drws, 2021-09-17:
Before any other convincing I have to ask why the buffer content wouldn't be writable at all? I'm all for read-only option, maybe also by default, but why would modification of any buffer be a high risk if the application allows or even requires it?

I use CircularBuffer to store structs representing found nodes and there is a need to update buffer elements with new information. I've solved this by using the CircularBuffer to only store addresses (offsets) of an ordinary array, providing me with a complete circular buffer logic while also being able to directly access the data.

Another example is DSP. CircularBuffer could provide a flexible sample pipeline where the signal is processed in-place. Also, writable buffers can be useful in any kind of filtering.

rlogiacco, 2021-09-17:
Ok, the DSP in-place processing actually makes sense. What I am concerned about though is you might also want to use the array-like write operation to add/remove elements from the buffer, which is something I don't want to support: that is what I consider a high-risk operation and once you have array-like write you might complain _why can't I just write in position 50 if my buffer has a capacity of 100?_

You can't imagine how many absurd questions I get on this library.

Maybe a write operation can return a boolean reporting if the operation was successful and it was pointing to a valid position.... but I'm pretty sure 95% of the time the users will just ignore the return value....

rlogiacco, 2021-10-04:
Any thought on my considerations @drws ?

drws, 2021-10-04:
I agree with what you said and I was also not talking about modifying the buffer structure, just the data, in-place. Since you marked it as an enhancement I thought no further comments were necessary. I'm also all for write return status as you described.

rlogiacco, 2021-10-05:
Are you willing to submit a PR with your changes, a usage example and some documentation in the README?

drws, 2021-10-05:
> I've solved this by using the CircularBuffer to only store addresses (offsets) of an ordinary array, providing me with a complete circular buffer logic while also being able to directly access the data.

I haven't tried to fix CB since currently I have a multi-year project finishing in following months (hobbies @ minimum) and I've been able to quickly implement all of the logic needed as described above.

I am however interested in making CB a go-to library after that - and for which write operations would be required. But to be honest with you, I still haven't decided whether CB's memory protection is a blessing or a curse. I'm a C-guy, I don't mind memory being corrupted. Yes, that is a joke :D

closed 2021-11-03 by github-actions[bot]