rlogiacco/CircularBuffer #51: Overwriting pointers to objects and having memory leaks

Dim0N22, 2021-09-24:
The doc says:
> The very same applies for the pop() and shift() operations as any dynamically allocated object is only detached from the buffer, but the memory it uses is not automagically released (see the Object.ino example)
> ```cpp
> Record* record = new Record(millis(), sample);  // a dynamically allocated object
> buffer.push(record);
> 
> // somewhere else
> if (!buffer.isEmpty()) {
>     Record* current = buffer.pop();
>     Serial.println(current.value());
>     delete current; // if you don't do this the object memory is lost!!!
> }
> ```

So it's clear that we need to manually clear memory for dynamically allocated objects. Now due to this being a circular buffer, a pointer to an object can be overwritten when we do 
```cpp
buffer.push(record);
```
in an already full buffer. In this case `buffer.push()` will return `false`, but the pointer and therefore the memory will be lost.
What we probably could do is returning the overwritten object instead of `false`, this way consumers will be able to do whatever they need with the object.

The other option, of course, is to check whether the buffer is full before doing `push` but that's not very convenient in my opinion.

Or am I missing something and there are better approaches for this?
Thanks.

rlogiacco, 2021-09-24:
This makes sense, but what would you return to indicate no overwrite has occurred? 
Keep in mind we are dealing with a buffer aiming to cover both simple and complex data types, as well as compile-time and run-time allocated structures...

rlogiacco, 2021-10-04:
In other words, if the `push()` method returns an instance of `T`, what would you return to indicate _no T_ in a language having no typed definition of `null`?
If the buffer is on unsigned integers, which value would you return to indicate _no unsigned integer has been overwritten_?

I understand the need for checking if the buffer is full might be counterintuitive to newcomers and programmers with a Java or similar language background, but I don't see a clear and non misleading alternative course of action in C/C++. I'm open to your opinion and suggestion, but I need clarification.

rlogiacco, 2022-01-18:
This issue has remained stale for over 3 months: it's time to close it.

closed 2022-01-18 by rlogiacco