rlogiacco/CircularBuffer #58: Writing a single data fills the buffer

pkrssy, 2021-12-05:
v1.3.3
If a single data is written into the newly created buffer, buffer.size() reports one but all the other elements of the buffer is filled with the data. 

#include <CircularBuffer.h>
CircularBuffer<long, 6> buffer;
uint16_t data1 = 1111;

void setup() 
{

  Serial.begin(9600);
  Serial.print("Buffer size before writing: ");
  Serial.println(buffer.size());
  buffer.push(data1);
  Serial.print("Buffer size after writing: ");
  Serial.println(buffer.size());
  Serial.print("Elements: ");
  Serial.print(buffer[0]);
  Serial.print(", ");
  Serial.print(buffer[1]);
  Serial.print(", ");
  Serial.print(buffer[2]);
  Serial.print(", ");
  Serial.print(buffer[3]);
  Serial.print(", ");
  Serial.print(buffer[4]);
  Serial.print(", ");
  Serial.println(buffer[5]);
   
}

void loop() {
  
}

Result:

09:28:44.723 -> Buffer size before writing: 0
09:28:44.856 -> Buffer size after writing: 1
09:28:44.889 -> Elements: 1111, 1111, 1111, 1111, 1111, 1111

The behavior can not be experienced in v1.3.0

rlogiacco, 2021-12-05:
Any access beyond the current buffer bounds is a violation of the contract and assuming anything about the buffer internal state is a usage error.
If the values returned for the populated buffer positions is correct then the buffer operates as intended.

rlogiacco, 2021-12-06:
Another way to describe the behaviour, if you want to print the buffer contents you should only print the items between position `0` and position `buffer.size() - 1`: in your case between `0` and `0` after the first insertion....

See the two for-loops examples at https://github.com/rlogiacco/CircularBuffer#automatic-optimization and the code at https://github.com/rlogiacco/CircularBuffer/blob/cf4ded93a099eb77c9451ee56b3eeaf31c69b7a3/examples/Test/Test.ino#L6-L27 for reference.

If you want to inspect the internals for debugging reasons, you can enable debugging and use the two `debug` functions exposed by the library:

```cpp
#define CIRCULAR_BUFFER_DEBUG
#include <CircularBuffer.h>
CircularBuffer<long, 6> buffer;
```

And then you can call `buffer.debug(Serial)` or `buffer.debug(Serial, customFunction)` to see the whole buffer contents with the head and tail positions.

closed 2021-12-06 by rlogiacco

pkrssy, 2021-12-06:
You're right but for a simple user like me it would be more comfortable to indicate the data is not available (return "nan") without the need to add additional conditions in a for cycle.

rlogiacco, 2021-12-07:
I’m not aware of the existence of such type.
I would throw an exception, but that’s also unavailable: I’m open to
suggestions which can be safely and soundly I mplemented on a
microcontroller.