rlogiacco/CircularBuffer #59: CircularBuffer in RTC Memory while DeepSleep on ESP32

EdMlt, 2021-12-29:
Hi,

I was trying to save my circular buffer in the RTC memory of my ESP32 while it is in DeepSleep. So I tried this simple function where I have struct and a circular buffer that are modified before going to Deep-sleep.
The struct is modified, the ID_msg is incremented every time but the buffer size is still 1 even is I push a new value at every wake up:

```c++
RTC_DATA_ATTR sensorsRecord newrecord{20, ID_sensor, ID_msg, epochUnion, ValueUnion, intValue, 30};
RTC_DATA_ATTR CircularBuffer<sensorsRecord, 50> test_SensorsDatas;

void test_RTC_memory2(){
    newrecord.ID_msg.value++;
    ID_msg.value++;
    Serial.println(String(newrecord.ID_msg.value));
    test_SensorsDatas.push(newrecord);
    Serial.println("buffer size: " + String(test_SensorsDatas.size()));
    esp_deep_sleep_start();
}```

So I don't know if it comes from my code or the library isn't compatible with RTC memory. Some advise or help would be appreciated. Thank you !

rlogiacco, 2021-12-29:
First of all, what I am going to say is based on a quick read of the specs of the ESP32 RTC memory: if you post your question anywhere with a higher number of viewers (like StackOveflow) you might get a more accurate answer.

I believe the library, as it is, is not compatible with RTC memory because while you are using the `RTC_DATA_ATTR` on the CircularBuffer instance, the same does not apply to the internal variables of the buffer.

My understanding is that you will have to add the `RTC_DATA_ATTR` modifier to all internal variables, and to test if my assumption is correct you can open the `CircularBuffer.h` file and add the modifier to these variables:
[image]

If that is the case and your test is successful, I can find a proper way to deliver this functionality in a new release.

EdMlt, 2021-12-29:
Thanks for your quick answer !
Yes I have posted the question on StackOverflow too: https://stackoverflow.com/questions/70520984/circularbuffer-in-rtc-memory-while-deepsleep-on-esp32
I tried to add it, it doesn't want to compile because this atribute is for when you want to create the buffer I think

rlogiacco, 2021-12-29:
For what I have seen, each piece of data that should end into RTCmemory has to be marked with that attribute and the CircularBuffer uses 4 different pieces of data: a pointer to the internal buffer, a pointer to the head, a pointer to the tail and a number counting the stored items. If any of those pieces are _lost_ then the whole CircularBuffer is unable to operate.

By marking the `test_SensorsDatas` as RTC I believe you will only be storing the pointer the instance in RTC memory, but I might be wrong. Let's see what SO says on this...

closed 2022-01-22 by github-actions[bot]