rlogiacco/CircularBuffer #60: Reading data in the historical order

strange-v, 2022-01-14:
First of all, neat library, thank you!

Do I understand correctly that currently, it is impossible to read values in the order, as they were added (without removing them from the buffer)? For example, if I log sensor values and then want to display them on a chart, I should read them from the most recent ones to the oldest.

Theoretically, it can work in this way:
```cpp
template <typename T, size_t S, typename IT>
void inline CircularBuffer<T, S, IT>::iterate(void (*callback)(T value))
{
	T *addr = tail;
	while (1)
	{
		callback(*addr);

		if (addr == head)
			break;

		*addr--;
		if (addr < buffer)
		{
			addr = buffer + capacity - 1;
		}
	}
}
```
But I'm not sure that everything is 100% correct, as the library uses more complicated code than I usually write on C++.

Do you think that adding such an iterator is useful?

strange-v, 2022-01-15:
A few additional comments.

1. A method to read in FIFO order also can be added.
2. Things that should be taken into account:
    - If we have a buffer of 100 values, the LIFO reading/iterating will return <= 100 values. Less than 100 in the case, when during the reading a new value was added.
    - In the same case, the FIFO reading will return >= 100 values.

In my particular case, I'll probably store a struct that contains a timestamp and a few values, so sorting can be performed on the client.

rlogiacco, 2022-01-16:
If you want to retrieve data without modifying the buffer in the same order as it was added, the code is simply:

```cpp
for (decltype(buffer)::index_t i = 0; i < buffer.size(); i++) {
    // do whatever you want with buffer[i]
}
```

The array-like accessor gives you read access to all the elements in the buffer, with index 0 being the head (the first element) of the buffer.

To iterate over the buffer as a FIFO... reverse the for-loop to access it as LIFO.

closed 2022-01-16 by rlogiacco

strange-v, 2022-01-17:
Thanks, @rlogiacco. I haven't noticed that in the current implementation, index 0 always points to the oldest record, neat!