rlogiacco/CircularBuffer #72: Namespace conflict with internal Arduino CircularBuffer

ubidefeo, 2022-11-17:
While trying to compile a Sketch downloaded from Barilla, 
I found that I could not compile it since it required `<RCircularBuffer.h>` which was nowhere to be found.
The error of a missing `index_t` type prompted me to do some research and found your library.

To make it work I had to rename the Class and finally got the sketch to compile.

Is there any chance to implement a namespace?

The Sketch which breaks is the following 
https://github.com/twofingerrightclick/BarillaPassiveCookingDevice/blob/main/Arduino/sketch_PastaThermometer_nano/sketch_PastaThermometer_nano.ino

The developer very likely did exactly what I did to make it work, rename Class and its occurrences in templates

rlogiacco, 2023-11-21:
As reported qin the issue you are referencing, the correct solution is to use a specific reference to the library.

The sketch you are referencing should be changed so that no class gets renamed: if the author did, he/she should have included the renamed class.

With regards to namespaces, I don't see namespaces being used in the microcontroller world... can you give me examples of libraries using namespaces widely adopted in the microcontroller domain, and thus change my percpeption?

Thanks a million!

rlogiacco, 2023-11-21:
In addition there's a potential fix in a dedicated branch `fix/library-include` which I was waiting for validation: can you please verify if that's of any help?

closed 2023-12-15 by github-actions[bot]